CrazyRedMachine/LUFAHybridFightstick #2: Rumble question

Seikishiki, 2021-05-02:
Hi! i wanted to know if there is a way to check the rumble/force feedback from the usb port and the reports? I wanted to be able to launch an action from that check. I'm a beginner in this field so i wanted to ask for advice on the feasability of this... Thank you for your time!

CrazyRedMachine, 2021-05-02:
hi, it depends on what you mean by launch an action. the firmware is aware of rumble data, so you can setup some action in the firmware such as lighting a led or powering an actual rumble motor you put inside the controller..

for this you'd have to research what the rumble reports look like and then write some code inside that function to check if the report is for rumble and in which case perform the wanted actions : https://github.com/CrazyRedMachine/LUFAHybridFightstick/blob/c676b4ecc4a08c5241b20825dd1154e2d6c83b97/XS_HID.c#L42

if you mean launching an action on the computer itself it's more annoying cause when a program sends gamepad rumble data, the other programs are not aware of it afaik... in your case if you're a beginner your best bet would probably be to use a virtual joystick app, not sure if some of them support mapping actions to rumble data, but general googling might help for this.

Seikishiki, 2021-05-03:
Thank you for your answer!!!

I wanted to be able to do something like that https://github.com/Linyoa/Enhanced-Training-Mode : I need to get the rumble activation info (For smash ultimate you get a rumble when you get hit) from the nintendo switch and launch some action on the pad.

Thanks a lot for your answer! I have to learn more about hid and reports to be able to do that then! If you have any god documentation or advice on how to check and research the report structure or just hid beginner stuff i will gladly take it :D

CrazyRedMachine, 2021-05-03:
welcome, here is a basic but very useful tutorial on hid reports https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/ (this is how I first learnt on the subject).

output reports are the same thing except they come from the computer towards the device.

about checking the report struct, the thing you have to check is the hid descriptor in the code. (or you can use a soft like "simplehidwrite" (you can download it on ianaxelsson page) and see what it sees when you plug a real nintendo switch controller on your pc

CrazyRedMachine, 2021-05-03:
the plot thickens.. the switch controller report in this firmware is, as usual, based on the hori pokken controller, but it seems this controller model doesn't rumble, so there's the possibility that you'll have to modify the hid descriptor part of the code as well.. i'd definitely check both this firmware and a real pro controller and try to spot differences in their descriptors (you can use "tdd" to dump descriptors)

this is the switch descriptor in the firmware :
https://github.com/CrazyRedMachine/LUFAHybridFightstick/blob/c676b4ecc4a08c5241b20825dd1154e2d6c83b97/Descriptors.c#L10

closed 2021-06-30 by CrazyRedMachine