CrazyRedMachine/LUFAHybridFightstick #3: How to reflash board?

going2town96, 2021-06-12:
I'm using a sparkfun qwic pro micro and the upload worked but I got the pinout wrong for my particular project and I noticed there's no COM port to upload new code anymore. How do I flash new code into the board?

CrazyRedMachine, 2021-06-12:
sorry it's dead :(

nah j/k, there is a pin labeled RST, what you have to do is briefly make that pin go in contact with GND when the arduino ide is looking for the board (you can wire an arcade button between RST and GND or since they are adjacent you can just prepare a flat screwdriver or something, click upload on the arduino ide, then once it's done compiling and shows "Uploading..." you use the screwdriver to make rst and gnd touch for one sec then you remove it, the board should then start flashing and the code will be uploaded).

If that doesn't work, you can try the second method : keep RST and GND in contact, click upload, when the IDE says "uploading", you release contact, then put in contact again briefly and release again. ( basically if it were with a button it would be "hold.... release push release")

going2town96, 2021-06-12:
I tried that, didn't work. I noticed my board has its own reset button, then found out on the board documentation that you have double tap the reset button to put the board into bootloader mode. So all I had to do was tap reset twice while it's uploading. Sorry for not checking my board documentation first.

closed 2021-06-12 by going2town96