CrazyRedMachine/LUFAHybridFightstick #5: SOCD cleaning L3+R3 (question)

ItaloKnox, 2021-06-29:
Just came across this repo while doing some research. I'm receiving a couple of Pro Micros later today, so I want to give this code a try, even though I'm a total newbie at arduinos (literally my first time).

I seem to grasp the concept pretty well, and it looks pretty easy to pull off, but I'm having trouble figuring out which pin is the L3/R3 mentioned in the README. Looks like the SOCD modes are triggered by doing a combination of L3+R3+something else, but I can't really find my answer in the code.

Also, this part looks a bit confusing to me:
https://github.com/CrazyRedMachine/LUFAHybridFightstick/blob/ef6d77c7c7a5618fb21e15b3224b5985fcd32a15/LUFAHybridFightstick.ino#L13-L15

Does that mean that it will always do SOCD cleaning regardless of on-the-fly configuration being on or off?

CrazyRedMachine, 2021-06-29:
L3/R3 are named LS and RS in the pinout section... i also just noticed that pin 18 is actually A0 so it looks like there is a conflict in the default pinout... and pin 17 is not exposed on the pro micro unfortunately. but you can always change the pinout at will.

about socd cleaning, yes, it will always perform socd cleaning, what can be enabled/disabled is "changing the socd mode" if that makes sense

ItaloKnox, 2021-06-29:
That's what I get from shutting off my brain while looking for something. Makes sense, thanks!

As for the pins, I guess I'll have to look into enabling RX1/TX1 for the LS/RS buttons, but that's on me from now on.

closed 2021-06-29 by ItaloKnox

ItaloKnox, 2021-06-29:
Hey! Sorry to use this issue for an issue unrelated to the scope of the repo, just wanted to know one thing about the boards: is it normal for the COM port to stop showing up after I upload the code? The arduino is fine and works, but if I want to change the code, I need to connect GND to RST and then release the RST while uploading the code to COM1. After that, it changes to COM3 and is successfully updated. Just wondering if that's normal behavior or if there's any workaround for easier uploading after the first one.

CrazyRedMachine, 2021-06-29:
yes this is because LUFA allows you to define the usb descriptors, and this firmware only implements xinput or pokken controller descriptors. The serial interface is not implemented (neither the Nintendo Switch nor the Microsoft XInput driver would recognize the controller if you were to add it back)

ItaloKnox, 2021-06-29:
Makes total sense, I'll add a button to my project for easy resets. Once again, thank you. 👍

CrazyRedMachine, 2021-06-29:
welcome :)  (do you really plan on reflashing the firmware often once the controller is working? with the pro micro i just use either a piece of dupont cable or a screwdriver to put rst and gnd in contact if i have to reflash... )

ItaloKnox, 2021-06-29:
Not really, actually. But I'm planning to mount it into a perfboard and add male xh2.54 connectors for easy wiring (that way I can just use the zero delay cables I already own), so a button for resetting is not a big deal to add and might save me time if I decide to mount it.