CrazyRedMachine/LUFAHybridFightstick #7: Permission to use fork in product?

nullstalgia, 2021-07-28:
Howdy!

I'm also a fan of converting controllers. My specific way of going about it was a little different, though. I made a board to sit between a SNES/NES/Wii Classic Controller (with the Nunchuk connector, via I2C) and a Switch/PC. My original solution was to have the chip reflash itself with [miniboot](https://github.com/mihaigalos/miniboot) on every startup, but I found your project and was really impressed! You've managed to solve a lot of issues we've both run into (LUFA and Arduino mingling and mixing the XInput and Switch descriptors)!

I was able to design a whole new PCB around it, and they arrived today. But, I realized I should probably ask for an "OK" since you don't have a license in the repository (we might have to stick to GPLv3 due to bootsector/XInputPadMicro using it).

My repository is 

https://github.com/nullstalgia/NullWiiCon

My tindie page for them is

https://www.tindie.com/products/nullstalgia/nullwiicon/

I don't have the newest revision up on either of those yet, but I will very soon. (and I haven't sold a single one on Tindie yet. I was unhappy with the last PCBs so I kept the stock at zero)

Here's a couple pics of the newest board

[image]

[image]

CrazyRedMachine, 2021-07-28:
Hi,

First at all nice work on your pcb, looks great.

I don't mind it using a fork of my firmware as long as credit is given (but I see you're already doing a good job with what you've been using so far), and your project stays open source (software+hardware) as well (but once again seems like it's the case).

By the way I see you have a toggle switch for xinput/switch directly on the board, so just in case you haven't seen the "PCFX" branch of this firmware (i took advantage of a toggle switch on the controller), know that you can hotswap between the descriptors as well (without having to physically unplug/replug) :

https://github.com/CrazyRedMachine/LUFAHybridFightstick/blob/a9c0dc80aee243b11b502921be18a76bdea3a5b1/LUFAHybridFightstick.ino#L144

Oh and while it's definitely not necessary, having one board to play with would be much appreciated :P

Wishing you best of success with this board

nullstalgia, 2021-07-28:
Yeah, I watch the state of the pin and if it doesn't match the current mode it runs this:

```cpp
    wdt_enable(WDTO_250MS);
    while (true)
      ;
```

To start the chip over from scratch.

I can absolutely send you one! Only if you promise to test my firmware and give feedback ;)

Email me at my github name at gmail. ;)

---

Thanks for the good wishes and compliments. I haven't seen the other branches extensively yet, will do.

Cheers!

CrazyRedMachine, 2021-07-29:
sure, of course i'll test it that's the whole point :D
thanks a lot

oh yea, resetting the whole board is a bit hardcore but why not :P the
thing is you can call USB_disable() and then call the usb setup function
again to force reenumeration (and disabling/reenabling interrupts properly
since lufa uses them to process usb packets)

nullstalgia, 2021-07-30:
Despite my scope having a screenshot feature, I didn't want to move 3 feet for a USB drive

On my firmware, I have the TX LED toggle its state every time `Endpoint_Write_Stream_LE` is called.

This is a reading on my older firmware, built around mid 2020.

[image]

And after upstreaming it.. I thought it was just not being called since it looked solid. But when I shook it, I saw that it was actually just going double the speed!

[image]

But the Switch limits the speed to 65 "hz" for both versions. I thought I had a picture, but nope.

I'm going to use it for a little bit to see if anything breaks.

If you want to peek at the PCB and my new code, it'll be here before it'll be on Main.

https://github.com/nullstalgia/NullWiiCon/tree/V2.2

CrazyRedMachine, 2021-08-01:
this is weird, 65Hz is pretty slow... the switch should be polling the controller at 250Hz by default (I changed the descriptor to get a 1000Hz polling rate as it seems it's still working this way even on a real console, PC does poll at 1000 in this case, but not sure about the switch..)

nullstalgia, 2021-08-04:
Well every rising and falling edge each is a full cycle. Quick maths says 65*4 = 260, which isn't far off 250. Am I extrapolating info out of thin air?

closed 2024-01-05 by CrazyRedMachine