CrazyRedMachine/LUFAHybridFightstick #9: Can't upload code anymore

SamimanX, 2021-09-06:
Hi! I was trying to make an arcade stick with this code and an Arduino pro micro, I uploaded the code and my pc recognizes the Arduino as a Xbox Controller, but when I was doing the wiring I realized that many of the pins listed in the .ino doesn't exist in this pcb, so I tried to change the pins in the .ino file but when I tried to upload the code again I noticed that the COM port that initially worked to upload the code dissapeared, so I can't upload the changes do I say goodbye to my Arduino? Thanks in advance for any help :)

CrazyRedMachine, 2021-09-07:
you should read the arduino-lufa readme... when using lufa you lose the arduino serial interface, so in order to reflash you have to use the reset button to put the bootloader in DFU mode.

on the pro micro there's a RES pin, if you join res and gnd that's like pressing the reset button. do that briefly while the arduino ide is looking for the device (and shows "uploading..." ) then it will find it and reflash

SamimanX, 2021-09-07:
Thanks for the answer :) Shortly after I posted this question saw the same thing in the closed issues and tried with the reset button, tried pushing 2 times, holding it before the "uploading" and releasing it, tried all the mentioned methods there with no success, the only thing I think I would be failing it's about the timing, in some forums saw that the timing it's tight.

CrazyRedMachine, 2021-09-07:
the "hold.. release push release" method is what has always been working for me.

programmer should be the default one (nothing selected here).

also go in file preferences, check "show verbose output during : compilation and upload"

now when you click upload, after compilation you should see message going {COM}->{COM}
 while it's looking for the arduino.. this is the moment at which you're supposed to hit the reset button, then a com port number will appear and the board will be flashed

SamimanX, 2021-09-07:
Yeah, using the "show verbose" option at least shows me that reseting the arduino starts something, but now its giving me the "Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding" error, already tried with a blank sketch and changing USB port, the cable is brand new and nothing :/

CrazyRedMachine, 2021-09-07:
Here is an output example for me (progammer is "AVRISP mkII" on this computer but I doubt this makes a diff, i'm just uploading directly to the arduino through usb) :

```
Sketch uses 15710 bytes (54%) of program storage space. Maximum is 28672 bytes.
Global variables use 760 bytes (29%) of dynamic memory, leaving 1800 bytes for local variables. Maximum is 2560 bytes.
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}                                      << at this point I put reset and gnd together then release
PORTS {} / {COM20, } => {COM20, }
Found upload port: COM20
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega32u4 -cavr109 -PCOM20 -b57600 -D -Uflash:w:C:\Users\redacted\AppData\Local\Temp\arduino_build_830370/sdvxController.ino.hex:i 
```

SamimanX, 2021-09-07:
Solved the problem, the issue was that I'm using a homemade PCB to be able to connect al the wiring without soldering, I was doing the reset button out of it, but taking out the arduino and doing the reset with a screwdriver let me upload the sketch again, so moral of the story, don't have anything connected on the pins when doing this.

[image]

closed 2021-09-07 by SamimanX

CrazyRedMachine, 2021-09-07:
oh, it can definitely still work with stuff connected... I wired a small microswitch and drilled a hole through my neogeo pad shell so I can reset by inserting a small tip.. rest of the buttons are always soldered to the arduino

most importantly, glad you solved the issue :)