CrazyRedMachine/LUFAHybridFightstick #12: SOCD last input input problem

epiccpete, 2022-02-11:
I've set my SOCD to LAST_INPUT and I'm having a problem where if I press and hold Left, then down, then right, and release down, it goes back to left instead of continuing on to right. Is there an easy way to fix this? The motion I'd like to do is hold left and then tap down and then right and get a half circle back to forward motion for Guilty Gear Strive

CrazyRedMachine, 2022-02-12:
hi,

i gave the code a quick read and it seems like it should be working properly as far as clean_socd() goes, however the socd state change might be a little hard to trigger it seems, so just to make sure, you can try the following

https://github.com/CrazyRedMachine/LUFAHybridFightstick/blob/4497b245b819b18700de2b84728811332de14d04/LUFAHybridFightstick/LUFAHybridFightstick.ino#L108

put = LAST_INPUT for both variables there, then comment out the EEPROM.get for socd on lines 222 and 223, this should ensure you are indeed in last input mode...

if that doesn't work, please try your motion in all directions (so left, down right and release down,   right down left and release down,  down right up and release right, etc and tell me if there's a similar issue or not)

thanks,

CrazyRedMachine, 2022-02-13:
thanks for the details, I think I understand what's happening here, you can try the firmware from this branch : https://github.com/CrazyRedMachine/LUFAHybridFightstick/tree/socd_fix

epiccpete, 2022-02-13:
That has fixed it.. Truly appreciate your help, thank you :)

CrazyRedMachine, 2022-02-13:
glad to hear this, i'm gonna merge it to the main code :) and thank you very much for your appreciation :D

closed 2022-02-13 by CrazyRedMachine