CrazyRedMachine/LUFAHybridFightstick #14: Error compiling for board Arduino Leonardo (LUFA).

D4vv1-1d, 2022-02-16:
Hi, I have an error saying  Error compiling for board Arduino Leonardo (LUFA) and I don't  understand what is wrong because I am shaw I followed the orders

CrazyRedMachine, 2022-02-16:
you should enable verbose output during compilation and flash in the arduino ide preferences.
then you check what errors you have in the bottom frame. "compilation error" by itself gives no useful info for anyone to help you..

make sure you installed the Bounce2 lib as well. Also try to compile the code as is at least once before adding your changes to verify your install is ok

D4vv1-1d, 2022-02-16:
ok i will try that

D4vv1-1d, 2022-02-16:
this is the fool error 

Multiple libraries were found for "Descriptors.h"
 Used: C:\Users\OneDrive\Documents\Arduino\libraries\LUFAHybridFightstick
In file included from C:\Users\AppData\Local\Temp\arduino_modified_sketch_590784\LUFAHybridFightstick.ino:1:0:
 Not used: C:\Users\OneDrive\Documents\Arduino\libraries\LUFAHYBRIDLibary
C:\Users\OneDrive\Documents\Arduino\libraries\LUFAHybridFightstick/Descriptors.h:7:10: fatal error: LUFA/LUFA/Drivers/USB/USB.h: No such file or directory
 Not used: C:\Users\OneDrive\Documents\Arduino\libraries\ATMega32U4-Switch-Fightstick-master
 #include <LUFA/LUFA/Drivers/USB/USB.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Leonardo (LUFA).

D4vv1-1d, 2022-02-16:
it works now thanks

D4vv1-1d, 2022-02-16:
Because there is no joystick.h