CrazyRedMachine/LUFAHybridFightstick #15: Wiring

D4vv1-1d, 2022-02-19:
for the joystick do I just wire in the x axis and y axis or do I need to wire the 5v for the joystick and wire the ground for the buttons?

CrazyRedMachine, 2022-02-19:
of course you have to power the potentiometer so that will require a gnd and 5v... (and of course the pot you're using has to support 5v...). buttons also need ground but it's irrelevant for the x/y analogs.. you might want to watch/read tutorials about padhacks and how gamepads work etc, that will answer most of your questions

D4vv1-1d, 2022-02-19:
Ok

closed 2022-02-19 by CrazyRedMachine

D4vv1-1d, 2022-02-19:
(it's irrelevant for the x/y analogs) do you mean I don’t wire the VER / Y and the HOR / X?

CrazyRedMachine, 2022-02-19:
no, i just mean x and y axis are not buttons. once again please realize that github issues are meant to report bugs and issues about this code, it's not a tech support channel. I've taken note and I might add wiring diagrams to this repo so that it's easier for beginners, but I'm quite busy with other projects for now. In the mean time for general questions like this you might try the cons&stuff discord for example ( https://discord.gg/TKSNREy ). Thanks

D4vv1-1d, 2022-02-19:
Sorry thanks for the help I will now use this discord and thanks in future for when you make a diagram

CrazyRedMachine, 2022-02-19:
there are also plenty of tutorials available online about analog sticks with Arduino, that will show you how they are meant to be plugged and will explain the general principle