CrazyRedMachine/LUFAHybridFightstick #16: Switch not recognizing controller

VenomFriend, 2022-02-27:
Hello, I've tried your code after trying the original one from fluffymadness. Just like his, your code works like a charm when I connect my controller to my PC, and even on my android phone with a USB adapter. But no matter what I do, my friend's god damn switch won't recognize the controller. I've tried it on both usbs in the docker and the small USB in the switch, using the same adapter that I used to connect to my phone, but nothing. The LED on the board turns on, so it's being powered, but it's not being recognized as a controller. 

I'm using a Pro Micro board, so I'm wondering if the problem is that this board in particular isn't recognized on the switch? His switch is updated to the latest firmware/version, and the option to use USB pro controllers or something like that is turned on as well. And in your code in particular I've selected the POKKEN CONTROLLER on my PC before connecting it to the switch, just in case. Even tested to see if all buttons are being recognized. The only change I had to do was with the order/mapping of the pins, to match my board and the order that I used when I tried with fluffymadness' version. I've also tried using both "Arduino Micro(LUFA)" and "Arduino Leonardo(LUFA)" options, just in case... Both worked normally on my PC, but no changes on the switch.

CrazyRedMachine, 2022-02-27:
hi, this is weird as I did try this code on my switch and it's working correctly there (and the board inside my neogeo pad2 is a pro micro as well)...

is the switch a regular one, a lite or an oled, also what is the exact firmware version? maybe I can find someone with the same model to test... I also suggest testing on another friends' switch if you can find other models (although that would be sad if they indeed decided to block some controllers in later revisions).

VenomFriend, 2022-02-27:
I've managed to make it work. What finally did the trick was to use the option on the switch to forget/delete all paired controls, and then reboot the console. Kinda weird that this did the trick, but I ain't complaining. Anyway, thanks for your code! Tthe SOCD that you wrote is really useful, and much better than the one I had poorly done when I used fluffymadness' code.

closed 2022-02-27 by VenomFriend