CrazyRedMachine/LUFAHybridFightstick #17: How to calibrate joystick

D4vv1-1d, 2022-04-09:
Which app do I enter (persistent calibration mode) and how do I do it

CrazyRedMachine, 2022-04-09:
it's written in the readme https://github.com/CrazyRedMachine/LUFAHybridFightstick#real-analog-support

basically hold both analog clicks (keep them centered tho) while plugging the controller, then start doing a bunch of circles so that the arduino can detect the min and max values of each axis from each stick... the values are saved to memory on the fly, so after a bunch of rotations you can unplug the controller and plug it back normally, previously stored values will remain

D4vv1-1d, 2022-04-09:
Do I need any apps open or do I just plug and do and does the controller have to be Xinput mode?

D4vv1-1d, 2022-04-09:
Or can it be in Nintendo switch controller mode cause when I tried doing what you did it didn’t work but I am not certain if I didn’t do enough circles

CrazyRedMachine, 2022-04-09:
that's written in the firmware and should work in any mode... no app is needed. what issue are you having with your analog sticks exactly ?

D4vv1-1d, 2022-04-09:
I try calibrate it and when I put it one the switch it don’t work

D4vv1-1d, 2022-04-09:
And when I curcle them do they still need to be clicked down

CrazyRedMachine, 2022-04-09:
no you don't need to keep pressing the analogs as you're making the circles. But if you get absolutely no response from the analogs then you probably have a wiring issue. Calibrating is just to adjust the range, even without any calibration you'd still get signal.
Make sure the analogs are wired to analog pins, and that you uncommented the line there 

https://github.com/CrazyRedMachine/LUFAHybridFightstick/blob/2309fdde4d49283c16e2e28bbf8016510a10ca55/LUFAHybridFightstick/LUFAHybridFightstick.ino#L14

to enable real analog support.

D4vv1-1d, 2022-04-09:
By uncommented do you mean delete?

CrazyRedMachine, 2022-04-09:
i mean remove only the "//" at the beginning of the line.

D4vv1-1d, 2022-04-10:
Hi I’ve done what you said but now in the IDE when I was uploading it says  No dead zone set

CrazyRedMachine, 2022-04-10:
then guess what to do with that line...

https://github.com/CrazyRedMachine/LUFAHybridFightstick/blob/2309fdde4d49283c16e2e28bbf8016510a10ca55/LUFAHybridFightstick/LUFAHybridFightstick.ino#L15

D4vv1-1d, 2022-04-10:
uncommitted it

CrazyRedMachine, 2022-04-10:
uncomment it and it'll work yea.. in C starting a line with // makes it a comment, so it's not part of the code anymore... I'm using this to enable/disable options. WITH_ANALOG requires DEADZONE to be set. Once both of these are back in the code, the analog handling functions will be active.

D4vv1-1d, 2022-04-11:
Hi I got the code working but when I tried to upload it it couldn’t find my Arduino so it wouldn’t let me change the code so how do reset my Arduino  from this code

CrazyRedMachine, 2022-04-11:
use the reset button or reset pin.

D4vv1-1d, 2022-04-11:
My Arduino Leonardo i s reading itself as a poken controller and no longer a Arduino

D4vv1-1d, 2022-04-11:
Do I hold it?

CrazyRedMachine, 2022-04-11:
you can hold it then release it once the ide starts looking for the arduino to flash it.
please refer to the countless tutorials you can find about that everywhere on the internet.

D4vv1-1d, 2022-04-11:
It didn’t work

closed 2022-04-29 by CrazyRedMachine