CrazyRedMachine/LUFAHybridFightstick #19: Programmatically trigger button

bobbbob, 2022-05-12:
I was hoping you could point out to me the functions that I could use to trigger a button programmatically. I’ve attempted a few ways I thought would work but to no avail.
My background is in ObjC so possibly it’s my lack of understanding in some key differences that’s holding me back.
I am looking for the ability to trigger the action of a button pressed without the need for a physical button. For example when another condition is met within an if statement that’s in the main loop.

CrazyRedMachine, 2022-05-12:
hi,
just have to set values in the buttonStatus array... 0/1 for digital inputs, 0-255 for analogs

for example buttonStatus[BUTTONX] = 1; or buttonStatus[AXISX] = 200;

if it's dpad values then you need to set internalButtonStatus[] instead so it can then be converter to the proper mode.

bobbbob, 2023-01-08:
Sorry, are you able to quote a quick sample of what the loop() would look like to trigger a button. I have tried your code directly in the loop and I have tried editing the buttonRead() just to test.

CrazyRedMachine, 2023-01-08:
you need to do it after the call to buttonRead (since this is what builds the buttonStatus array) so for example at line 291, 

buttonStatus[BUTTONX] = 1;

then it will act like the x button is always pressed..

bobbbob, 2023-01-08:
Thank you for your fast reply, I’ve done exactly as you have said and it still won’t work for me. I even tested on a directional as this would be more visibly evident when held down.

closed 2023-01-08 by bobbbob