CrazyRedMachine/LUFAHybridFightstick #20: How to Reflash?

anitokyomics, 2022-07-22:
Hi. 

I would like to know how to reflash the arduino pro micro with the LUFAHybridFightstick? 

Because I have edited the pinout to make it work for my configuration as a test and didn't know if I could flash it again to default or whatnot.

CrazyRedMachine, 2022-07-22:
yes you can reflash, you just need to bring RES and GND pins together for a short while (it's the reset button) when the Arduino IDE is looking for the board to upload the firmware

anitokyomics, 2022-07-22:
Oh nice! 

Thank you very much. I am able to reflash it. :)

anitokyomics, 2022-07-22:
Ah nevermind. Please disregard my last comment. Everytime I short it to reset, it disconnects when I start to upload it.

Any solution to ensure that I can upload the code?

CrazyRedMachine, 2022-07-22:
of course reset will disconnect the board, but then during the first couple seconds the ide will be able to find the board in bootloader mode. you have to reset when the code is compiled and the ide is looking for the board to flash it.

i suggest enabling verbose output during compilation and upload in the ide preferences to be able to see better when to hit the reset button.

anitokyomics, 2022-07-22:
Alright. I am now able to reflash. That tip helped me a lot.

SamuelRamirezDP, 2022-08-21:
I added a function for stopping the HID from starting if a specific input is detected for easy reflashing, is this commonly requested? should I push this function?

anitokyomics, 2022-08-21:
I think it would be best to add it if ever there will be future updates or if there are some screw-ups when editing the pinout mapping.

CrazyRedMachine, 2022-08-21:
hid is not the problem though, I thought the problem is the lack of cdc endpoint with magic packet handling? what makes it easier for you to flash in your case?
tbh i'm not sure I want to document another button combination when the reset button exists..

anitokyomics, 2022-08-21:
The reset button is ok regardless. I am still able to flash it with the reset button but it takes a bit of "when" to press the reset button to let the Arduino IDE upload the code to the device.

closed 2024-01-05 by CrazyRedMachine