CrazyRedMachine/LUFAHybridFightstick #22: Left analog moves left and right stick at the same time.

Makushimo, 2024-01-05:
Hello, I enabled the analogs, I connected a magnetic sensor (MLX90333) that replaces the potentiometers; I connected it to pins A0 and A1. The problem is that the sensor moves the left and right lever at the same time. I would only need the left lever. Thank you
(I used Google translator, I hope you can understand me)

CrazyRedMachine, 2024-01-05:
nothing is plugged on A2/A3 ? then it will pickup random noise.

you can force right analog to center by adding
```
buttonStatus[AXISRX] = 128;
buttonStatus[AXISRY] = 128;
```
right after line 288, after `axisRead();` and before `#endif`

Makushimo, 2024-01-05:
There is nothing connected to A2/A3.
**It worked!** Thank you very much for your help and your great work.

CrazyRedMachine, 2024-01-06:
welcome, glad it's working :D enjoy!

closed 2024-01-06 by CrazyRedMachine