CrazyRedMachine/LUFAHybridFightstick #23: I need help. Can the vertical analog axis be inverted?

Makushimo, 2024-01-11:
Hello again, here I am, assembling the lever (copy of
Sanwa) with the magnetic sensor (MLX90333), due to the position of the sensor, when I move the physical lever **up**, the output signal is **down**, that is, I need to invert the **Y** axis. The horizontal analog or **X** axis is correct.
Sorry for the inconvenience, I'm very new to Arduino.
Thank you. Greetings!
(translated by Google)

CrazyRedMachine, 2024-01-11:
axis value is just a value between 0 and 255.

if you want to invert it just replace value by 255-value.

therefore, same place where you added the other instructions to disable right analog before, you can also add:
`buttonStatus[AXISLY] = 255 - buttonStatus[AXISLY];`

Makushimo, 2024-01-11:
Thank you very much!, that worked!
One more doubt. Can I calibrate the left lever with the right lever locked? I connect buttons to A4 and A5, and hold them down when connecting the Arduino, then move the left stick in circles. But I'm not sure if it works.

CrazyRedMachine, 2024-01-11:
it should work fine, each stick is calibrated independently (it's just trying to detect the voltage range and the center position of each analog), you just need to wire the r3 button so you can boot in calibration mode.

You can check after calibration if analog stick responds fine with gamepadtester or joy.cpl etc..

Makushimo, 2024-01-11:
It works almost well, I think it's a hardware issue, the magnet describes a circular path or almost like a rhombus on the magnetic sensor, and that path is transferred to the output.
The bad thing is that, in the GamepadTester graph, the cardinal directions do not reach 100%, they are at most 92%, I think it is acceptable.
In a few days, I will get the box to assemble the controller, I am eager to try it in a game.
Thank you very very much.

CrazyRedMachine, 2024-01-11:
maybe i should add a config tool to dump and update calibration data manually.. i'll see if i can find time to do it this weekend, not sure.

i hope it will work well with your games :)
how are you setting the magnetic sensor in place ?

Makushimo, 2024-01-11:
A friend designed and printed a kind of 3D bridge for me, to place the sensor, that bridge was a little low, I had to raise the height with some cardboard so that the magnet, glued to the bottom of the lever, does not touch the sensor .
[image]
[image]