CrazyRedMachine/LUFAHybridFightstick #24: Regarding the SS pinout

AM-SOR, 2024-06-10:
Hello,

I apologize if this isn't the right way to ask about code issues, but I'm new to Arduino coding. I'm trying to create a fightstick using the provided code. However, I'm having trouble with the SS pinout. I've researched but still can't figure out where to solder for the SS pinout. Could someone please help me with this?

Thank you!

CrazyRedMachine, 2024-06-11:
depends on the arduino you're using. on leonardo there are two vias besides the tx and rx leds. ss is on rx led (the via closer to the usb port). that being said, this code doesn't require it

AM-SOR, 2024-06-11:
I am actually trying to create a hitbox of sorts and I have read somewhere your code has Xinput support. I am also using a Arduino "pro" micro.

CrazyRedMachine, 2024-06-11:
yes it has xinput support, won't work on a real xbox360 or xbox one though since there's additional encryption stuff on consoles.

pro micro doesn't have many pins, but i guess you won't add real analog sticks to your hitbox so just recycle the analog pins at your convenience, you just have to change them in the pinout.h define

CrazyRedMachine, 2024-06-11:
to any number which is not already used by another pin... and since you don't have analog joysticks, you can use any of the 4 analog pins that are normally used by them

AM-SOR, 2024-06-18:
Ohhh I see. Thanks for the help. I do have more inquiries if you don't mind me asking? It's about combined inputs.

CrazyRedMachine, 2024-06-18:
always better to ask than to ask to ask x)

AM-SOR, 2024-06-18:
Just making sure. I do not wish to waste your time to be honest with these questions. But I was wondering since my hit box has 3 buttons for home start and back. But I lack buttons for the RS and LS. I wish to do a combined input for the sticks. For example LB and Y would be the left stick while RB and B would be the right stick. Again I apologize since arduino coding is very new to me and I do not wish to brick my board.

CrazyRedMachine, 2024-06-19:
starting from this line
https://github.com/CrazyRedMachine/LUFAHybridFightstick/blob/2309fdde4d49283c16e2e28bbf8016510a10ca55/LUFAHybridFightstick/LUFAHybridFightstick.ino#L577

you can check button pressed by checking buttonStatus[].. as you can see the code right there is checking for start+select together.

then if you want to force a button press, just write buttonStatus[START]=1; for example 

don't worry you won't be able to brick your board by writing bad code

AM-SOR, 2024-06-19:
So under that line can I add a command to have combined inputs for the right stick and the left stick?

CrazyRedMachine, 2024-06-19:
yes.. something like this

if (buttonStatus[BUTTONSELECT] && buttonStatus[BUTTONLB]) {
 buttonStatus[BUTTONLS] = 1;
/* if you want to disable select and lb inputs when this happens */
 buttonStatus[BUTTONSELECT]=0;
 buttonStatus[BUTTONLB]=0;
}

AM-SOR, 2024-06-19:
I see now. Thank you for your time and help.